lbernstone/arduino-esp32-solo #7: Task watchdog gets triggered

atban98, 2020-08-25:
I am using the ESP32-SOLO module. The program I am running:
`void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}`

the module is resetting with the following error:

`rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
E (10325) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10325) task_wdt:  - IDLE0 (CPU 0)
E (10325) task_wdt: Tasks currently running:
E (10325) task_wdt: CPU 0: loopTask
E (10325) task_wdt: Aborting.
abort() was called at PC 0x400d1537 on core 0

Backtrace: 0x4008a0a8:0x3ffbe370 0x4008a2b5:0x3ffbe390 0x400d1537:0x3ffbe3b0 0x40083bce:0x3ffbe3d0 0x400e62ab:0x3ffb1fb0 0x400874dd:0x3ffb1fd0

Rebooting...
`
Can you help me with this one, please?

Kashyapkoshti, 2020-08-28:
rst:0x10 (RTCWDT_RTC_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 

I am getting this error in my esp32 devkit v1
How to solve it?
I am using arduino IDE for programming the esp32

lbernstone, 2020-09-21:
These errors have nothing to do with this library.  Search for solutions in the arduino-esp32 issues.

closed 2020-09-21 by lbernstone

s00500, 2021-06-17:
For anyone searching for this later: https://github.com/espressif/arduino-esp32/issues/5048