lbernstone/arduino-esp32-solo #10: Arduino core for the ESP32 - 2.0.0

bfaliszek, 2021-11-09:


bfaliszek, 2021-11-09:
And 2.0.1 RC1

bfaliszek, 2021-11-09:
it would be more convenient to create a tutorial on how to generate the necessary files by yourself

JoJ123, 2021-11-10:
@bfaliszek Did you found already a solution?

bfaliszek, 2021-11-10:
not yet. For now, I'm back to Arduino core 1.0.6

bfaliszek, 2021-11-15:
2.0.1 - Arduino core for the ESP32 https://github.com/espressif/arduino-esp32/releases/tag/2.0.1

sixtyfive, 2021-11-15:
Perhaps even more convenient would be to get support for the single-core chips into the core :-)

bfaliszek, 2021-11-19:
Officially, support for ESP32-SOLO-1 has been added to the roadmap for version 2.2.0 - https://github.com/espressif/arduino-esp32/milestone/4

ChuckMash, 2022-02-03:
Anyone have a workaround yet?

lbernstone, 2022-03-14:
I have posted libraries for 2.0.2, but don't have a solo around to test it on.  If I get confirmation that the examples all build (especially bluetoothserial), then I'll make a release.
You can build the libraries for yourself using [lib-builder](https://docs.espressif.com/projects/arduino-esp32/en/latest/lib_builder.html).  The only thing that needs to be changed in sdkconfig is CONFIG_FREERTOS_UNICORE=y

lbernstone, 2022-03-26:
Pretty sure this is working now, so I will close the issue.

closed 2022-03-26 by lbernstone