lbernstone/arduino-esp32-solo #11: Question about arduino-esp32-solo

astrowalker2013, 2022-03-27:
Hi, 
Thanks for your work, but we have issue to compile in order to used your library for shelly device who are embeded esp32 unicore.

we developed an OS for ESP32 and ESP8266, our project compile and run on different device. 
we developed with ARduino ide 1.8.15 in windows 10.
This OS used a lot of
The Compilation are OK  before used your library

we are upgrade in 2.0.2 for esp32 device and replace by  your file,   but the compilation doesn't OK.

The error display :

```
c:/users/xxxxxxxxxxl/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\xxxxxxxxxxl\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32/lib\libesp_littlefs.a(esp_littlefs.c.obj):(.literal.esp_littlefs_init+0x50): undefined reference to `xPortEnterCriticalTimeout'
c:/users/xxxxxxxxxxl/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\xxxxxxxxxxl\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32/lib\libesp_littlefs.a(esp_littlefs.c.obj): in function `vPortEnterCritical':
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/esp_littlefs/src/esp_littlefs.c:325: undefined reference to `xPortEnterCriticalTimeout'
```

What could be the issue ? It could be a missing library ?

lbernstone, 2022-05-17:
@Jason2866 Can you change the README.md on your repo to show a platformio.ini env using your release?

Jason2866, 2022-05-18:
@lbernstone  Like this [https://github.com/Jason2866/esp32-arduino-lib-builder/blob/main_idf441/README.md](https://github.com/Jason2866/esp32-arduino-lib-builder/blob/main_idf441/README.md)